bulletphysics/bullet3 #3328: Air Friction

1234shel, 2021-03-31:
I am trying to model an underwater environment. I think it should be possible by modifying gravity and air friction. The question is does pybullet support air friction? Also is there a better of way of modelling an underwater environment in pybullet?

erwincoumans, 2021-04-10:
There is no air model built-in. There is some air model for soft body, but only exposed in C++ (see the Example Browser)

closed 2021-04-10 by erwincoumans

Lu-Meichen, 2023-11-29:
> 我正在尝试对水下环境进行建模。我认为通过改变重力和空气摩擦力应该是可能的。问题是 pybullet 是否支持空气摩擦？另外，有没有更好的方法可以在pybullet中对水下环境进行建模？

Did you succeed to model an underwater enviroment? I want to do it based on pyBullet, but I do not get enough useful sources. Could you help me?

1234shel, 2023-11-29:
I think you have to play around with the gravity parameter for this.