bulletphysics/bullet3 #3329: Question about CPU memory usage

seann999, 2021-04-01:
TL;DR does taking 150MB RAM for one YCB object look right?

Hi, I'm currently using PyBullet to train an RL agent.
I'm using multiple parallel environment instances to speed up training, but I'm currently limited by how much memory each instance uses. After checking the system monitor as the environment loads, it seems like most of the memory consumption comes from the objects (URDFs and SDFs) being loaded in.
Most of the objects are from the YCB dataset (google16k), and loading 30 of them takes up over 5GB. The raw files for each object (like .objs and .pngs) add up to more or less 10MB, but when loaded into the physics simulation they seem to take 5GB/30= about 150+MB each. Does that sound right?
I'm using collision meshes generated by V-HACD.

If the numbers look more or less right and there is no easy way to optimize memory usage, I might look into simply upgrading the hardware.

erwincoumans, 2021-04-01:
Hard to tell without an example file (urdf + obj + texture) + simple script to load it.

seann999, 2021-04-01:
@erwincoumans 
Here is a gist to load the models, but the models have to be downloaded separately, as written at the top of the script... If this is a problem, please tell me
https://gist.github.com/seann999/bb01271dc2d20c84ed1495e08e025887

erwincoumans, 2021-04-10:
The textures seem very big when expanded (they are stored compressed), and stored at least twice (once for regular OpenGL, another time for TinyRenderer). Then the vertices are bigger in memory as well, but that has much smaller effect.
I would recommend using lower resolution textures.

closed 2021-04-10 by erwincoumans