bulletphysics/bullet3 #3334: About using VR in PyBullet

1210652636, 2021-04-02:
I do not know how to use VR in PyBullet, I even do not know how to start. Please help me. I am very confused about it.

cadop, 2021-04-02:
Hopefully these resources can help

1. Page 72 of the pybullet quickstart guide https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3
1. Examples folder has a few VR things https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/examples
1. This kind of sparse doc (referenced in the quickstart guide) https://docs.google.com/document/d/1I4m0Letbkw4je5uIBxuCfhBcllnwKojJAyYSTjHbrH8/edit#

1210652636, 2021-04-06:
Thanks a lot! I have known these three links and I even tried the examples about VR, but it didn't work, the program can not connect to my VR and I do not know why. I didn't find a very detailed tutorial to help me connect to VR.

BluePrintRandom, 2021-04-06:
blender xr_action branch is wrapping openXR

I think window_manager.c has the relevant code to run a XR session

1210652636, 2021-04-06:
[image]
Now I can only run an empty VR world and there is nothing in this world. I can not see the robot or any other items in the world. And I run the "https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/examples/vrEvent.py", nothing happened, I even can not see any output in the output page.
[image]

erwincoumans, 2021-04-06:
Are you running App_PhysicsServer_SharedMemory_VR?
Once that runs fine, and you see an empty world, I recommend try running vr_kuka_setup.py next:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py

1210652636, 2021-04-06:
[image]
I performed a series of operations based on what is shown in the image above.
[image]
The script generates a series of files. Actually I did not see "App_PhysicsServer_SharedMemory_VR.exe", I saw files with the name "App_PhysicsServer_SharedMemory_VR" in other formats, which I then opened in MSVC.
I try to run the "App_PhysicsServer_SharedMemory_VR":
[image]
[image]
Right-click on "App_PhysicsServer_SharedMemory_VR" and run it. All I can see is the following image.
[image]
I run other codes that give the following screen, this may not have anything to do with VR.
[image]

And then I try to run "https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py", and nothing happened. The steps I ran were as follows: 1. Right-click on "App_PhysicsServer_SharedMemory_VR" and run it and I would see a blank screen in VR (with nothing in it but two handles). And then I run "https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py" and it just runs there by himself, nothing is output, nothing happens, and nothing is visible in VR.
[image]
Am I doing the steps correctly? I know maybe I am a little stupid but I am try my best to do these things, so pls tell me about this, thanks a lot!!

1210652636, 2021-04-06:
> Are you running App_PhysicsServer_SharedMemory_VR?
> Once that runs fine, and you see an empty world, I recommend try running vr_kuka_setup.py next:
> https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py
> 
> Best to run a plain terminal and regular python (3) version, no ipython or other versions.

image
I performed a series of operations based on what is shown in the image above.
image
The script generates a series of files. Actually I did not see "App_PhysicsServer_SharedMemory_VR.exe", I saw files with the name "App_PhysicsServer_SharedMemory_VR" in other formats, which I then opened in MSVC.
I try to run the "App_PhysicsServer_SharedMemory_VR":
image
image
Right-click on "App_PhysicsServer_SharedMemory_VR" and run it. All I can see is the following image.
image
I run other codes that give the following screen, this may not have anything to do with VR.
image

And then I try to run "https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py", and nothing happened. The steps I ran were as follows: 1. Right-click on "App_PhysicsServer_SharedMemory_VR" and run it and I would see a blank screen in VR (with nothing in it but two handles). And then I run "https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py" and it just runs there by himself, nothing is output, nothing happens, and nothing is visible in VR.
image
Am I doing the steps correctly? I know maybe I am a little stupid but I am try my best to do these things, so pls tell me about this, thanks a lot!!

1210652636, 2021-04-06:
> Are you running App_PhysicsServer_SharedMemory_VR?
> Once that runs fine, and you see an empty world, I recommend try running vr_kuka_setup.py next: [...]

[image]

I performed a series of operations based on what is shown in the image above.

[image]

The script generates a series of files. Actually I did not see "App_PhysicsServer_SharedMemory_VR.exe", I saw files with the name "App_PhysicsServer_SharedMemory_VR" in other formats, which I then opened in MSVC.
I try to run the "App_PhysicsServer_SharedMemory_VR":

[image]

Right-click on "App_PhysicsServer_SharedMemory_VR" and run it. All I can see is the following image.

[image]

I run other codes that give the following screen, this may not have anything to do with VR.

[image]

And then I try to run "https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py", and nothing happened. The steps I ran were as follows: 1. Right-click on "App_PhysicsServer_SharedMemory_VR" and run it and I would see a blank screen in VR (with nothing in it but two handles). And then I run "https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vr_kuka_setup.py" and it just runs there by himself, nothing is output, nothing happens, and nothing is visible in VR.

[image]

Am I doing the steps correctly? I know maybe I am a little stupid but I am try my best to do these things, so pls tell me about this, thanks a lot!!

erwincoumans, 2021-04-06:
You need to leave that "App_PhysicsServer_SharedMemory_VR running of course. Also, it needs to compiled in 64bit mode.

The run that vr_kuka_setup.py from a regular Python (64bit) terminal:

python3 path/to/vr_kuka_setup.py

Note that the pybullet plugin needs to ge up-to-date.
Ot should connect using SHARED_MEMORY.

1210652636, 2021-04-07:
> You need to leave that "App_PhysicsServer_SharedMemory_VR running of course. Also, it needs to compiled in 64bit mode.
>
> The run that vr_kuka_setup.py from a regular Python (64bit) terminal:
>
> python3 [...]

Thanks for your patient and I try your method but it still didn't work.

[image]

[image]

It seems that the program doesn't work anymore at line 7：
[image]
[image]

My python version is 3.9.4：

[image]

Again, thanks for your patient.

erwincoumans, 2021-04-07:
It is about the version of pybullet, not of Python.
Best is to do this:
pip uninstall pybullet
then in the root of your installation run
python setup.py install
(replace with pip3 and python3 if needed)

1210652636, 2021-04-08:
omg, it works thanks a lot!!!

I think my mistake is that I forget to change "Win32" to "x64", so this is why it can not run the code.

[image]

1210652636, 2021-04-08:
> Make sure the executable name on Windows for the VR server is App_PhysicsServer_SharedMemory_VR_vs2010_x64_release.exe
> [image]
> When generating the visual studio project files with premake, it expect Python installation to be located in c:\python***
> so it can find the right include and library files.
> 
> It is about the version of pybullet, not of Python.
> Best is to do this:
> pip uninstall pybullet
> then in the root of your installation run
> python setup.py install
> (replace with pip3 and python3 if needed)

omg, it works thanks a lot!!!

I think my mistake is that I forget to change "Win32" to "x64", so this is why it can not run the code.

[image]

erwincoumans, 2021-04-22:
Great, glad it works. Closing.

closed 2021-04-22 by erwincoumans