bulletphysics/bullet3 #3335: Error getting View matrix and Projection matrix

shiyani21, 2021-04-02:
[demo1.zip](https://github.com/bulletphysics/bullet3/files/6249540/demo1.zip)

I am trying to get the view matrix and projection matrix and i keep getting this error:

pybullet build time: Feb 22 2020 01:54:01
b3Warning[examples/SharedMemory/PhysicsDirect.cpp,1052]:
requestOpenGLVisualizeCamera failedTraceback (most recent call last):
File "demo1.py", line 18, in
viewMat = p.getDebugVisualizerCamera()[2]
pybullet.error: Cannot get OpenGL visualizer camera info.

shiyani21, 2021-04-02:
[cube_obj.zip](https://github.com/bulletphysics/bullet3/files/6249641/cube_obj.zip)

cadop, 2021-04-02:
I don't get an error. What OS are you using? 

I do get 0's on the `viewMat` when using your example, but I suspect that is because you are running `p.DIRECT` instead of `p.GUI`.  If I change to `p.GUI`, 

```
viewMat = p.getDebugVisualizerCamera()[2]
print(viewMat)
```

outputs:

`(0.642787516117096, -0.4393851161003113, 0.6275069713592529, 0.0, 0.766044557094574, 0.36868777871131897, -0.5265407562255859, 0.0, -0.0, 0.8191521167755127, 0.5735764503479004, 0.0, 2.384185791015625e-07, 2.384185791015625e-07, -5.000000476837158, 1.0)`

Check the example for rendering which is set as direct mode, which creates a camera view. 

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/testrender_np.py

erwincoumans, 2021-04-10:
PyBullet uses OpenGL matrices, check somewhere else for details about OpenGL / rendering matrices.

closed 2021-04-10 by erwincoumans