bulletphysics/bullet3 #3336: Wrong transforms from View and Projection matrix

shiyani21, 2021-04-02:
[demo.zip](https://github.com/bulletphysics/bullet3/files/6249563/demo.zip)

I have a function read_obj that reads the vertices from the obj file 
then it goes into transform function that does 
v' = Projection_mat*view_matrix*model_matrix*vertex_pos(a 1x4 vector where the last value is 1)
this to get the vertex location from the world to camera coordinates
and the resulting transforms are:
the transfroms for vertices are [3.29889629 0.75614495 1.88041992 0.55808138]
this is wrong because the last value should be 1 and not 0.55808138 and then i would get the vertices after i normalize the vector by dividing by the 3rd value.

any clues as to how i can fix it?

shiyani21, 2021-04-02:
[cube_obj.zip](https://github.com/bulletphysics/bullet3/files/6249643/cube_obj.zip)

erwincoumans, 2021-04-22:
It is an OpenGL question, pybullet provides a method to compute both projection and view matrix. You can compute the model matrix, using getBasePositionAndOrientation and convert to matrix 4x4.

closed 2021-04-22 by erwincoumans