bulletphysics/bullet3 #3343: Lidar (rayTestBatch) can't detect holes inside an STL object

PierreExeter, 2021-04-07:
Hi,

I am trying to detect collisions with an object using `rayTestBatch()` and I was able to run the [example provided](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/batchRayCast.py) without issue.

I am now replacing the obstacle object by a circular widow - created from an STL and URDF. The widow frame is detected correctly as an obstacle, however the rays going through the window also detect the presence of an obstacle, see screenshot attached. I'm expecting the debug lines to turn green when no obstacle is detected.
[image]

How can I modify the behaviour so that the rays going through the window don't detect any object?
I attached a minimal working example.

[test_lidar.zip](https://github.com/bulletphysics/bullet3/files/6273540/test_lidar.zip)

Many thanks

erwincoumans, 2021-04-07:
You need to create a convex decomposition of the object, for example using vhacd (which is part of pybullet).
First convert from stl to obj, then run vhacd.
See also https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_examples/vhacd.py

The quickstart guide has some parameters of vhacd if needed.

closed 2021-04-07 by erwincoumans

PierreExeter, 2021-04-08:
It works perfectly, thanks a lot!