bulletphysics/bullet3 #3346: About the robot arm drops naturally

1210652636, 2021-04-09:
Pls help me. I am having some problems with the robot arm. (The urdf file for the robot arm was written with my own reference). When I move the robot arm to a position (tilt), I no longer care about the robot arm and the robot arm moves down on its own over time.
[image]
[image]
We can see the UR5 robot on the left moving slowly on its own, which is an interesting phenomenon and how I should have avoided it.
This is my "UR5.urdf":
`<?xml version="1.0" encoding="utf-8" ?>

<robot name="ur5">

    <material name="Lake_blue">
        <color rgba="0.8 1 1 1"/>
    </material>

    <material name="Light_grey">
        <color rgba="0.866 0.866 0.866 1"/>
    </material>

    <material name="Dark_grey">
        <color rgba="0.502 0.502 0.502 1"/>
    </material>

    <link name="base_link">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/base.stl"/>
            </geometry>
            <material name="Light_grey"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/base.stl"/>
            </geometry>
        </collision>
    </link>

    <link name="shoulder_link">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/shoulder.stl"/>
            </geometry>
            <material name="Dark_grey"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/shoulder.stl"/>
            </geometry>
        </collision>
    </link>

    <joint name="joint0" type="revolute">
        
        <origin xyz="0 0 0.0862" rpy="0 0 0"></origin>
        <axis xyz="0 0 1"/>
        <parent link="base_link" />
        <child link="shoulder_link"/>
        
        <limit effort="30" velocity="1.0" lower="-6.28" upper="6.28" />
    </joint>

    <link name="upperarm_link">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/upperarm.stl"/>
            </geometry>
            <material name="Lake_blue"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/upperarm.stl"/>
            </geometry>
        </collision>
    </link>

    <joint name="joint1" type="revolute">
        
        <origin xyz="0 0.136 0" rpy="0 0 0"></origin>
        <axis xyz="0 1 0"/>
        <parent link="shoulder_link" />
        <child link="upperarm_link"/>
        
        <limit effort="30" velocity="1.0" lower="-2.09" upper="0.523" />
    </joint>

    <link name="forearm_link">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/forearm.stl"/>
            </geometry>
            <material name="Lake_blue"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/forearm.stl"/>
            </geometry>
        </collision>
    </link>

    <joint name="joint2" type="revolute">
        
        <origin xyz="0 -0.12 0.425" rpy="0 0 0"></origin>
        <axis xyz="0 1 0"/>
        <parent link="upperarm_link" />
        <child link="forearm_link"/>
        
        <limit effort="30" velocity="1.0" lower="-1.57" upper="1.57" />
    </joint>

    <link name="wrist1_link">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/wrist1.stl"/>
            </geometry>
            <material name="Dark_grey"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/wrist1.stl"/>
            </geometry>
        </collision>
    </link>

    <joint name="joint3" type="revolute">
        
        <origin xyz="0 0 0.3922" rpy="0 0 0"></origin>
        <axis xyz="0 1 0"/>
        <parent link="forearm_link" />
        <child link="wrist1_link"/>
        
        <limit effort="30" velocity="1.0" lower="-3.14" upper="1.57" />
    </joint>

    <link name="wrist2_link">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/wrist2.stl"/>
            </geometry>
            <material name="Dark_grey"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/wrist2.stl"/>
            </geometry>
        </collision>
    </link>

    <joint name="joint4" type="revolute">
        
        <origin xyz="0 0.093 -0.0015" rpy="0 0 0"></origin>
        <axis xyz="0 0 1"/>
        <parent link="wrist1_link" />
        <child link="wrist2_link"/>
        
        <limit effort="30" velocity="1.0" lower="-1.57" upper="1.57" />
    </joint>

    <link name="wrist3_link">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/wrist3.stl"/>
            </geometry>
            <material name="Light_grey"/>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="D:/Work/Internship/porject/object/meshes/wrist3.stl"/>
            </geometry>
        </collision>
    </link>

    <joint name="joint5" type="revolute">
        
        <origin xyz="0 0 0.0948" rpy="0 0 0"></origin>
        <axis xyz="0 1 0"/>
        <parent link="wrist2_link" />
        <child link="wrist3_link"/>
        
        <limit effort="30" velocity="1.0" lower="-6.28" upper="6.28" />
    </joint>

</robot>

`

1210652636, 2021-04-09:
[image]
The third picture.

erwincoumans, 2021-04-09:
>> When I move the robot arm to a position (tilt),

Are you using position control for all joints? (pybullet.setJointMotorControl2(robot, joint, pybullet.POSITION_CONTROL, pos,...)
See https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_examples/vr_kuka_setup.py#L46

closed 2021-04-09 by erwincoumans

1210652636, 2021-04-09:
> > > When I move the robot arm to a position (tilt),
> 
> Are you using position control for all joints? (pybullet.setJointMotorControl2(robot, joint, pybullet.POSITION_CONTROL, pos,...)
> See https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_examples/vr_kuka_setup.py#L46
> 
> Also, load the URDF using useFixedBase=True
> 
> and URDF files need elements, see the URDF for KUKA.

I just used mouse to move and control the ur5 robot and the robot will move down by itself.