bulletphysics/bullet3 #3347: collision point to closest mesh vertex

shivanimall, 2021-04-09:
Hello @erwincoumans, 

I have to convert the collision point (on say body B) returned by `getContactPoints` to the closest mesh vertex contained in body B's .obj file or as returned by `getMeshData`

using `getContactPoints` function to get collision points on body B (body B's model vertices are contained B.obj)
I have to transform these collision points from WCS in pybullet to the model's coordinate system such that the points lie in AABB of body B mesh vertices in B.obj

For this, I re-align the orientation using rotation matrix and subtract the current `base_position` returned by getBasePositionOrientation  at each timestep of collision from the collision point at that timestep.  Then translate it by values of AABB of body B.
The only concern is `getBasePositionOrientation` func returns the center of mass pos and not the geometric center of the object. Wondering if there is a more standard way to do this? or any pybullet function/support for this?

Let me know if my question is unclear, thanks!

erwincoumans, 2021-04-09:
If there the collision geometry has an offset, relative to the center of mass, you can get the local offset (position, orientation) transform using pybullet.getCollisionShapeData (see PyBullet Quickstart Guide)
Then you can multiply the transforms using pybullet.multiplyTransforms(base_pos, base_orn, colgeom_pos, colgeom_orn)

shivanimall, 2021-04-10:
thanks much Erwin, will try the above and update, in case I run into issues.

closed 2021-04-10 by erwincoumans