bulletphysics/bullet3 #3359: Forum registration not working

cadop, 2021-04-23:
Hi Erwin, 

I know you want to keep the issues as actual issues, and direct questions to the forum.  Registering on the forum is not working for me, and the forum keeps pointing to contacting board admins:
[image]

but the contact options are not available to people who are not logged in.  

-----

On a related note, I wonder if the github Discussions would work well for bullet.  https://docs.github.com/en/discussions 

I think if others are also having problems with the forum registration and coming to github for answers (then flooding the issues with questions instead of actual issues), the discussion board will at least offset that.  Likewise, its easy to move an issue into a discussion, so you could quickly clean up things that are not issues, without ignoring the question itself (and move discussions into issues, if appropriate). I also have been finding helpful answers in closed issues (more than the pybullet forums), but its not the easiest to search through.

_*again, sorry for posting a non-code issue in issues, but hoped its better than directly emailing you (and maybe its a meta-issue)._

tlemarec, 2021-04-23:
Having the same issue. Tried to open two different accounts, no success.

erwincoumans, 2021-04-26:
Thanks for the report, I made some mistake in configuring the server and email doesn't work at the moment. Need to find some time to fix that.

erwincoumans, 2021-04-26:
OK, I enabled 'discussions', but I can't promise to spend a lot of time on it (same for issues and forum).

closed 2021-04-26 by erwincoumans