bulletphysics/bullet3 #3811: Questions Regarding Collision Detection and the performCollisionDetection API not available for python

cmuell89, 2021-04-26:
I am trying to use PyBullet for performing motion planning research.

Essentially I would like to disable actual physical collisions that occur during simulation (since this can bump collision objects out of place) to test a variety of robot configurations for collision free validity. However, I have an issue in that I cannot seem to get robot object collision detection queries to work well.

Questions:
I've read in the documentation that getClosestPoints or getContactPoints should be called after calling performCollisionDetection however there is no available performCollisionDetection() function in the Python API. **How should I go about calling this function from the Python API? Is it even necessary to call if I use p.resetJointState to a test configuration and then call getClosestPoints() ?"**

**Does anyone have advice on how to perform collision queries for arbitrary arrangements of the robot and with static simulation objects?**

**Will using the p.setCollisionFilterGroupMask disable simulated collisions? Will it completely disable any form of collision detection?** I don't want to disable collision detection but I do want to disable collisions actually having physical consequences to my environment.

Here is the gist of how I am performing collision validity queries. Please note that some of the functions/objects are wrappers around PyBullet core functions such that I've created a little bit of a higher-level interface for my purposes:

I first create a Python Context Manager to disable self collisions of the robot and between the robot and any simulation objects so that when I reset the robot joint state, the arm does not slam into objects in the environment and displace them from their validity position for planning.  

```
class DisabledCollisionsContext(): 

    """
    Python Context Manager that disables collisions. Includes all self collisions, and collisions between SimObjects and Robots. 
    
    This does not disable collision checking but enables the robot to be repositioned into a configuration and then checked for collision without causing any physical simulation effects to occur.
    
    Attributes:
        simulator (simulator.Simulator): The Simulator singleton instance.
    """
    
    def __init__(self, simulator, excluded_bodies=[], excluded_body_link_pairs=[]): 
        """
        
        Args:
        simulator (simulator.Simulator): The Simulator singleton instance.  
        excluded_bodies (list): A list of PyBullet body IDs to excluded, generally reserved for SimObjects with only one link.
        excluded_body_link_pairs (list): A list tuples where the first element is the PyBullet Body ID and the second elemend is the link index. 
        """
        self.simulator = simulator
        self.excluded_bodies = excluded_bodies
        self.excluded_body_link_pairs = excluded_body_link_pairs
          
    def __enter__(self):
        p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,0) #makes loading faster
        self._disable_robot_collisions()
        self._disable_simobj_collisions()
        self.state_id = p.saveState()
      
    def __exit__(self, exc_type, exc_value, exc_traceback): 
        self._enable_robot_collisions()
        self._enable_simobj_collisions()
        p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,1) #re-enable rendering 
        p.restoreState(self.state_id)

    def _disable_robot_collisions(self):
        """
        Disables all self collisions for all robots.
        """
        for robot in self.simulator._robots.keys():
            if robot not in self.excluded_bodies:
                links = [idx for idx in range(-1, p.getNumJoints(robot))]
                for link in links:
                    if (robot, link) not in self.excluded_body_link_pairs:
                        p.setCollisionFilterGroupMask(robot, link, 0, 0)

    def _disable_simobj_collisions(self):
        """
        Disables all collisions between simobjects and robots.
        """
        for robot in self.simulator._robots.keys():
            for sim_obj in self.simulator._objects.keys():
                if sim_obj not in self.excluded_bodies:
                    p.setCollisionFilterGroupMask(robot, sim_obj, 0, 0)
[... 98 lines cut ...]

```
def robot_body_collision_test(joint_configuration, robot, object_body_id, client_id=0, max_distance=0.025):
    """
    Tests whether a given joint configuration will result in collision with an object. 

     It sets the robot state to the test configuration. There are no links test, we're testing whole body ids.

    Args:
        joint_configuration (list): The joint configuration to test for self-collision
        robot (int): PyBullet body ID.
        # link_pairs: 2D pairs of pybullet body links, in this case the pairs of potential in-self-collision pairs. 
        client_id (int): the physics server client ID.
    Returns:
        bool: True if no self-collision, else False.
    """
    robot_id = robot.get_simulator_id()
     # Set new configuration and get link states
    for i, idx in enumerate(robot._arm_dof_indices):
        p.resetJointState(robot._simulator_id, idx, targetValue=joint_configuration[i], targetVelocity=0, physicsClientId=client_id)
    if len(get_closest_points(client_id=client_id, body1=robot_id, body2=object_body_id, max_distance=max_distance)) == 0:
        return True
    else:
        return False
```

cadop, 2021-04-26:
If this is actually a question and not issue, it may be better for you to post it in the discussion as a Q&A category. 

https://github.com/bulletphysics/bullet3/discussions

Just my two-cents, since erwin did create it I think as part of keeping the issues clean, which I appreciated.

cmuell89, 2021-04-26:
@cadop That is fair. And i noticed very recently that he has included the performCollisionDetection call in a PR. I will close and move this to the discussion board. Thank you.

closed 2021-04-26 by cmuell89

erwincoumans, 2021-04-26:
performCollisionDetection would be a replacement for stepSimulation, to provide contact points, only for getContactPoints.
It is not needed for getClosestPoints.
The easiest is to keep a copy of a world, and synchronize their transforms and perform collision queries in the separate world. Have you considered that?

erwincoumans, 2021-04-26:
BTW, this discussion belongs in the Discussions section.