bulletphysics/bullet3 #3824: getLinkState returns nan

borninfreedom, 2021-04-30:
I used RLlib to train a robot, and the rllib returned the error
```
ValueError: ('Observation ({}) outside given space ({})!', array([ 0.6244375 , -0.14342715, -0.00830715,         nan,         nan,
               nan], dtype=float32), Box(-1.0, 1.5, (6,), float32))
```

The nan is from the code 
```python
    def _get_observation(self):
        self.object_pos = p.getBasePositionAndOrientation(self.object_id)[0]
        self.robot_pos = p.getLinkState(self.kuka_id, self.num_joints - 1)[4]
        observation = self.object_pos + self.robot_pos
        return np.array(observation).astype(np.float32)
```
So why getLinkState returns nan?

borninfreedom, 2021-04-30:
I run 10 envs parallel using rllib. But if I test the only one env not using rllib, it doesn't encounter the nan error.

erwincoumans, 2021-05-04:
It happens when you insert nan into pybullet.

closed 2021-05-04 by erwincoumans