bulletphysics/bullet3 #3825: The free energy density of the New-hooken model is wrong

ShoujingZheng, 2021-04-30:
https://github.com/bulletphysics/bullet3/blob/00cd39dc500dfe26818aaaae8db0ed5ac6a50fa2/src/BulletSoftBody/btDeformableNeoHookeanForce.h#L199

It should be:

		double density = 0;
		density += m_mu * 0.5 * (s.m_trace - 3.);
		density += (m_lambda * 0.5 +m_mu / 3.)* (s.m_J - 1. ) * (s.m_J - 1.);
		density -= m_mu * log(s.m_J );
		return density;

And the corresponding first piola stress and dP should also be modified.

ShoujingZheng, 2021-04-30:
the first Piola stress in line 358-363 should be:
	void firstPiola(const btSoftBody::TetraScratch& s, btMatrix3x3& P)
	{
		btScalar c1 = m_mu;
		btScalar c2 =(m_lambda +2. * m_mu / 3.)* (s.m_J - 1.) - m_mu / s.m_J;
		P = s.m_F * c1 + s.m_cofF * c2;
	}

ShoujingZheng, 2021-04-30:
the first Piola stress differential in line 367-375 should be:
	void firstPiolaDifferential(const btSoftBody::TetraScratch& s, const btMatrix3x3& dF, btMatrix3x3& dP)
	{
		btScalar c1 = m_mu ;
		btScalar c2 = 0;
		btScalar c3 = (((m_lambda +2. * m_mu / 3.)+m_mu /( s.m_J* s.m_J))* DotProduct(s.m_cofF, dF));
		dP = dF * c1 + s.m_F * c2;
		addScaledCofactorMatrixDifferential(s.m_F, dF, (m_lambda +2. * m_mu / 3.)* (s.m_J - 1.) - m_mu / s.m_J, dP);
		dP += s.m_cofF * c3;
	}

ShoujingZheng, 2021-04-30:
I modify the code according to :https://en.wikipedia.org/wiki/Neo-Hookean_solid 
[image]

erwincoumans, 2021-05-04:
Thank you, I'll try to get in touch with the original author. 

Do you mind creating a pull request in the meanwhile?

xhan0619, 2021-05-04:
@ShoujingZheng, the constitutive model implemented is a from [this paper from pixar](https://graphics.pixar.com/library/StableElasticity/paper.pdf), so the class name NeoHookean is somewhat misleading. 
The main motivation for adopting this model is that it doesn't have the log(J) term that is troublesome when the elements invert (which almost always happen with the size of the time step we are taking).

erwincoumans, 2021-05-04:
Moving to discussion. If you have inprovements to deformables, you can create a new issue or pull request.

closed 2021-05-04 by erwincoumans