bulletphysics/bullet3 #3975: Meaning of 'lateralFrictionDir1'and 'lateralFrictionDir2'

TneitaP, 2021-09-26:
In pybullet function "getContactPoints",  most of the returned value are belong to "bodyA" or "bodyB".  But for contact force, i.e. "**lateralFriction1**" & "**lateralFrictionDir1**" and  "**lateralFriction2**"  & "**lateralFriction2**" ， it is impossible to identify which body is subjected to the force from these variable names. 

My guess is that lateralfriction1 belongs to body A, and lateralfriction2  to body B. But if, according to this assumption, after all force on B are combined, **B is stationary** in the physics engine, but the combined external force (all normal forces, lateralfriction2 and gravity) **is not (0,0,0)**.

johbrust, 2021-09-28:
Hi, both forces act on both bodies (because of Newton's third law). Those forces act in the contact plane perpendicular to the contact normal. Therefore, `lateralFrictionDir1` and `lateralFrictionDir2` are orthogonal to the contact normal and to each other (e.g., calculated for deformable contacts at [this position](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBodyInternals.h#L1707)). The respective forces are the projections of the total friction force in the contact plane in those directions.

TneitaP, 2021-09-29:
Thanks for your patient reply. I also dived into the code of Bullet and found the clear explaination in <Bullet3Dynamics\ConstraintSolver\b3PgsJacobiSolver.cpp> : 

```
	void b3PgsJacobiSolver::setFrictionConstraintImpulse(b3RigidBodyData* bodies, b3InertiaData* inertias,b3SolverConstraint& solverConstraint, int solverBodyIdA, int solverBodyIdB, b3ContactPoint& cp, const b3ContactSolverInfo& infoGlobal)
	{
		b3SolverBody* bodyA = &m_tmpSolverBodyPool[solverBodyIdA];
		b3SolverBody* bodyB = &m_tmpSolverBodyPool[solverBodyIdB];

		{
			b3SolverConstraint& frictionConstraint1 = m_tmpSolverContactFrictionConstraintPool[solverConstraint.m_frictionIndex];
			if (infoGlobal.m_solverMode & B3_SOLVER_USE_WARMSTARTING)
			{
				frictionConstraint1.m_appliedImpulse = cp.m_appliedImpulseLateral1 * infoGlobal.m_warmstartingFactor;
				if (bodies[bodyA->m_originalBodyIndex].m_invMass)
					bodyA->internalApplyImpulse(frictionConstraint1.m_contactNormal * bodies[bodyA->m_originalBodyIndex].m_invMass, frictionConstraint1.m_angularComponentA, frictionConstraint1.m_appliedImpulse);
				if (bodies[bodyB->m_originalBodyIndex].m_invMass)
					bodyB->internalApplyImpulse(frictionConstraint1.m_contactNormal * bodies[bodyB->m_originalBodyIndex].m_invMass, -frictionConstraint1.m_angularComponentB, -(b3Scalar)frictionConstraint1.m_appliedImpulse);
			}
			else
			{
				frictionConstraint1.m_appliedImpulse = 0.f;
			}
		}

		if ((infoGlobal.m_solverMode & B3_SOLVER_USE_2_FRICTION_DIRECTIONS))
		{
			b3SolverConstraint& frictionConstraint2 = m_tmpSolverContactFrictionConstraintPool[solverConstraint.m_frictionIndex + 1];
			if (infoGlobal.m_solverMode & B3_SOLVER_USE_WARMSTARTING)
			{
				frictionConstraint2.m_appliedImpulse = cp.m_appliedImpulseLateral2 * infoGlobal.m_warmstartingFactor;
				if (bodies[bodyA->m_originalBodyIndex].m_invMass)
					bodyA->internalApplyImpulse(frictionConstraint2.m_contactNormal * bodies[bodyA->m_originalBodyIndex].m_invMass, frictionConstraint2.m_angularComponentA, frictionConstraint2.m_appliedImpulse);
				if (bodies[bodyB->m_originalBodyIndex].m_invMass)
					bodyB->internalApplyImpulse(frictionConstraint2.m_contactNormal * bodies[bodyB->m_originalBodyIndex].m_invMass, -frictionConstraint2.m_angularComponentB, -(b3Scalar)frictionConstraint2.m_appliedImpulse);
			}
			else
			{
				frictionConstraint2.m_appliedImpulse = 0.f;
			}
		}
	}
```

LFGfg, 2024-10-28:
> Those are two orthogonal friction directions. They are also orthogonal to the contact normal and lie in the separating plane between A and B. Friction doesn't 'belong' to a single body, it acts between two bodies.

Hi，How to understand lateralFrictionDir1 and lateralFrictionDir2, and how are they defined?