bulletphysics/bullet3 #3993: Two (potentially related) issues with restoring sleep state

mjlbach, 2021-10-13:
### Reproduction
The reproduction scripts, including assets, for the bug are located here:
https://drive.google.com/file/d/1-ym77X1yYny8EzQ-QRBPmW--U6NY4X2U/view?usp=sharing

The reproduction scripts are titled `bug_1.py` and `bug_2.py`.

### Issue 1 (bug_1.py)

`p.saveState` on a state containing awake objects does not result in the awake state being restored on `p.restoreState`. The state of an object on which you apply force must be restored manually by calling `p.changeDynamics` with the activation state *and* stepping physics with `p.step()`

### Issue 2 (bug_2.py)
In the same problem setup as 1:
* apply a force on object 3
* it hits object 2
* both objects go to sleep on the floor
* Teleport object 2 far away
* Let both objects go to sleep
* restoreState

After 4 restoreState calls, object 2 will no longer wake up as object 3 approaches.

erwincoumans, 2021-10-19:
Indeed. Save/restore state is not compatible with sleeping/deactivation at the moment. For now, it is best to disable sleeping if you rely on determinism and save/restore state.

erwincoumans, 2021-10-21:
You are welcome, and thanks a lot for clear description of reproduction of both bugs! I've been busy with [Brax](https://github.com/google/brax) and [tiny-differentiable-simulator](https://github.com/google-research/tiny-differentiable-simulator) projects, hope to sometimes come back to PyBullet to improve things like this.