bulletphysics/bullet3 #4023: Running a pybullet_env hangs if `.render("human")` is not called on OS X

bayerj, 2021-11-02:
Hey,

**tl;dr** Resetting a pybullet env without calling `env.render("human")` before enters an endless loop.

I am running on **Mac OS X Catalina, 10.5.7** and **Python 3.8**. I am installing pybullet in a fresh virtual env like this:

```
$ cd ~/tmp
$ mkdir pybullet_envs
$ mkvirtualenv -a pybullet_envs pybullet_envs
(pybullet_envs) $ pip install gym==0.21
(pybullet_envs) $ pip install pybullet==3.2.0
```

After that, I run the following simple script:

```
import gym
import pybullet_envs
env = gym.make('HopperBulletEnv-v0')
env.reset()
```

which produces the following output and then hangs:

```
pybullet build time: Oct 29 2021 12:21:45
/Users/bayerj/.virtualenvs/pybullet_envs/lib/python3.8/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
argv[0]=
```
I can then CTRL+Z the process and kill it:
```
argv[0]=
^Z
[2]  + 77905 suspended  python3 try_bullet.py
(pybullet_envs) ~/t/pybullet_envs ❯❯❯ kill 77905                                                                                                                                        ✘ 146
[2]  + 77905 terminated  python3 try_bullet.p
```

I can get it to run by explicitly setting render mode to `"human"` before reseting the env:

```
import gym
import pybullet_envs
env = gym.make('HopperBulletEnv-v0')
env.render(mode="human")
env.reset()
```

This does not work for `env.render(mode="rgb_array")`. 

Since I want to run the env in "headless" mode, this is unsatisfying. Ideally, the environment could be run without any rendering at all.

erwincoumans, 2021-11-03:
I can't reproduce the issue on a recent MacOS 11.6 version.
'''
import gym
import pybullet_envs
env = gym.make('HopperBulletEnv-v0')
env.reset()
'''
works fine.

bayerj, 2021-11-04:
Thanks @erwincoumans, I will update and see if the problem persists.

erwincoumans, 2021-11-06:
Thanks for the report. Please create a new issue if the problem persists.

closed 2021-11-06 by erwincoumans