bulletphysics/bullet3 #4025: Use C++17's `alignas` when __cplusplus>=201703L instead of per-compiler attributes in btScalar.h

devshgraphicsprogramming, 2021-11-03:
Hi,

We're trying to compile Bullet for Android x86 with Clang 11

When the compiler is Clang and the platform isn't win32, it does not go into any of the cases that define `ATTRIBUTE_ALIGNED16` as anything other than a passthrough.

It would provide some uniformity amongst all C++17 capable compilers, if alignments could be specified as `alignas()` instead.

erwincoumans, 2021-11-03:
pybullet is still c++2003 (with optional parts c++2011), you are supposed to add the preprocessor checks for your platform (android).

devshgraphicsprogramming, 2021-11-03:
okay so the code you compile might adhere to c++03 or c++11, but the compiler compiling might have had the `-std=c++17` or higher switch enabled.

All I'm asking is that you defer to your complex ladder of preprocessor checks only if `__cplusplus<201703L`

Does that somehow interfere?

erwincoumans, 2021-11-06:
That sounds reasonable, we could define ATTRIBUTE_ALIGNED16  using alignas for modern compilers. Do you mind creating a PR?

devshgraphicsprogramming, 2021-11-06:
I might eventually get around to it, yes

erwincoumans, 2021-11-12:
Thanks! Please consider a pull request if you get to it, closing this for now.

closed 2021-11-12 by erwincoumans