bulletphysics/bullet3 #4043: Get body acceleration info

raincold23, 2021-11-22:
For the development of legged robots, it will be greatly helpful if the controller could get acceleration info the floating base from simulator. Is it possible for users to get a function like getBodyAcc(bodyID) to retrieve body acc info from simulation?

bttner, 2021-11-26:
To my best knowledge, this is not possible. You need to keep track of the velocities and calculate the accelerations yourself, e.g.,
a(t) = (v(t) - v(t-1)) / dt, where v is the velocity and dt the sampling time (time step). The smaller your time step, the better the approximation. Note, however, that this is not the instantaneous acceleration and therefore not necessarily usable in controllers. 

The instantaneous acceleration could be calculated via f = m * a, where f is the acting force, m the mass and a the acceleration. This formula is used in Gazebo (search for Link::WorldLinearAccel in [linki](https://github.com/osrf/gazebo/blob/88afb773c10f5fd6399748da8ab334550df61431/gazebo/physics/Link.cc)). So, you would need to find out all acting forces on the desired link. But this information is only available for RigidBodies, i.e., https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Dynamics/btRigidBody.h provides getTotalForce(). Featherstone Bodies do not have this option for some reason, at least I can not find it. Thus, at the moment you need to find out all acting forces yourself if you want the instantaneous acceleration.

raincold23, 2021-12-02:
Thanks so much for your reply.

I cannot find the getTotalForce() method in pybullet document. Is that only for c++?