bulletphysics/bullet3 #4049: Simulate a 1D-Lidar (laser scan)

JiayunjieJYJ, 2021-11-25:
Can I simulate a 1D-Lidar (laser scan) by using the output of depth data of getCameraImage()? In other words, I want to get a horizonal distance(depth) data like a 1D-Lidar.  

Is it possible? Or is there any demo about it ??

Thank alot!!!

rohit-kumar-j, 2021-11-25:
Kindly put these queries [Discussions](https://github.com/bulletphysics/bullet3/discussions)
This is for a 3D point cloud... [pointCloudFromCameraImage.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pointCloudFromCameraImage.py). All you need to do is put it in the while loop for making the scan

JiayunjieJYJ, 2021-11-25:
> Kindly put these queries in [Discussions](https://github.com/bulletphysics/bullet3/discussions) This is for a 3D point cloud... [pointCloudFromCameraImage.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pointCloudFromCameraImage.py). You can put this in 2D as well. All you need to do is put it in the while loop for making the scan.
> 
> Idk why the video is so choppy
> 
>  Point.Cloud.mp4

Thanks! I will try it! Hope this method is accuracy enough.

rohit-kumar-j, 2021-11-26:
No problem! You could experiment with the resolution.

JiayunjieJYJ, 2021-11-27:
> No problem! You could experiment with the resolution.

Thank you! I made it!

closed 2021-11-27 by JiayunjieJYJ

JiayunjieJYJ, 2021-11-29:
> No problem! You could experiment with the resolution.

Hi, I find that if I run pybullet in DIRECT mode rather than GUI mode, the depth image can not detect the objects, do you know how to solve it?

rohit-kumar-j, 2021-12-01:
No, I have not faced this. Can you link the code here?