bulletphysics/bullet3 #4071: "pip install pybullet" and ""building wheel fails for bullet 3.2.1

rohit-kumar-j, 2021-12-04:
Error: building wheel fails, see also [Issues: #1864](https://github.com/bulletphysics/bullet3/issues/1864)

```
failed to build wheel for pybullet
```

rohit-kumar-j, 2021-12-04:
Solution: update Microsoft build Tools from [here](https://visualstudio.microsoft.com/visual-cpp-build-tools/)

closed 2021-12-04 by rohit-kumar-j

SrinjoyChoudhury, 2023-05-17:
> Solution: update Microsoft build Tools from [here](https://visualstudio.microsoft.com/visual-cpp-build-tools/), get bullet 3.2.1 release and extract from [here](https://github.com/bulletphysics/bullet3/releases/tag/3.21), extract the files
> 
> ```shell
> cd bullet3-3.21
> python setup.py build
> python setup.py install
> ```

Followed your instructions exactly. After `python setup.py build` I get this error:
error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/

I already have build tools installed (Version 17.6.0):
[image]

I get the same error if I continue to the next step.