bulletphysics/bullet3 #4112: CMake build fails on windows (MinGW-w64)

eLeCtrOssSnake, 2021-12-25:
Steps to reproduce:
1) Git clone bullet3
2) cd bullet3
3) mkdir build | cd build
3) cmake .. -G "MinGW Makefiles" -DBUILD_SHARED_LIBS=0 -DBUILD_EXTRAS=0
4) run ming32-make (or normal msys make)
5) get:
```
[ 46%] Linking CXX static library ..\..\lib\libBulletCollision.a
[ 46%] Built target BulletCollision
Makefile:144: recipe for target 'all' failed
mingw32-make: *** [all] Error 2
```

Previous versions have built fine previously. It seems like some minor cmake issue on windows, but I have zero knowledge in cmake.

Full build log: https://pastebin.com/bV2w1dwc

erwincoumans, 2022-01-04:
We use visual studio on Windows. For MinGW Makefiles you have to manually fix the build and consider sending a patch.

closed 2022-01-04 by erwincoumans

eLeCtrOssSnake, 2022-01-05:
That's a disappointing response for me. I've managed to build it by fiddling around with the checkboxes in cmake-gui. I might dig
deeper to find the real cause, and then (hopefully) submit a patch.

MirrorYuChen, 2023-08-22:
Meet the same problem，everything same to you, have you solved the problem?

eLeCtrOssSnake, 2023-08-22:
Yes, I just fiddled with some cmake flags and it worked.

MirrorYuChen, 2023-08-23:
Can you share me your methold? Thanks a lot!