bulletphysics/bullet3 #4114: Bullet Physics ExampleBrowser using openGL3+ [btgl] Release Build (Not Responding)

antonyjohne, 2022-01-02:
Hello!

I'm new to pyBullet and was testing out certain programs which ran as expected using the GUI command. However, after a while without making any changes to the code, the GUI browser started to freeze on startup and I had to force quit the browser. This issue persists for any code that I now un using the GUI command. Even a simple initialization program in pyBullet seems to crash/Not Respond the browser using the GUI command. I am using Windows and I even tried uninstalling the module and using the latest build straight from github, but the issue still persists. Here's a sample of code that I'm trying to run, but the browser still crashes (The code seems to work fine using the DIRECT mode, but I want to visualize the render). Any fixes would be greatly appreciated!

```
import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.GUI)

p.resetSimulation()
p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,0)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
planeId = p.loadURDF("plane.urdf")
p.setGravity(0,0,-10)

pendulum1Id = p.loadURDF(robot_dir, [0,0,1], [0,0,0,1], 0, 1, 0, 3.0)

p.getNumJoints(pendulum1Id)
#p.getJointInfo(pendulum1Id, 0)
#p.getJointState(pendulum1Id, 0)

maxForce = .001
mode = p.VELOCITY_CONTROL
p.setJointMotorControl2(pendulum1Id, 0,
controlMode=mode, force=maxForce)

p.setRealTimeSimulation(1)
p.disconnect()

```

OUTPUT:

```
pybullet build time: Jan  2 2022 09:06:35
starting thread 0
started testThreads thread 0 with threadHandle 00000000000001AC
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.14761 Compatibility Profile Context 21.30.22 30.0.13022.3
Vendor = ATI Technologies Inc.
Renderer = AMD Radeon(TM) Graphics
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 0000000000001E98
MotionThreadFunc thread started
```

erwincoumans, 2022-01-04:
Since you disabled rendering, you need to call 
```
p.configureDebugVisualizer(p.COV_ENABLE_RENDERING,1)
```
after loafing the urdf

closed 2022-01-04 by erwincoumans

mortal-thoughts, 2022-02-17:
how can I sovle the problem?