bulletphysics/bullet3 #4117: Memory corruption when enabling CCD in a multi threaded world

jrouwe, 2022-01-02:
When configuring a body to have a non 0 setCcdSweptSphereRadius and setCcdMotionThreshold in combination with a multi threaded world (btDiscreteDynamicsWorldMt) the following callstack corrupts memory:

btCollisionDispatcherMt::getNewManifold(const btCollisionObject * body0, const btCollisionObject * body1) Line 69	C++
btDiscreteDynamicsWorld::createPredictiveContactsInternal(btRigidBody * * bodies, int numBodies, float timeStep) Line 905	C++
btDiscreteDynamicsWorldMt::UpdaterCreatePredictiveContacts::forLoop(int iBegin, int iEnd) Line 101	C++
ParallelForJob::executeJob(int threadId) Line 122	C++
WorkerThreadFunc(void * userPtr) Line 371	C++
win32threadStartFunc(void * lpParam) Line 241	C++

At this time btCollisionDispatcherMt::m_batchUpdating is false, so multiple threads are racing on this piece of code:

manifold->m_index1a = m_manifoldsPtr.size();
m_manifoldsPtr.push_back(manifold);

This sometimes causes an incorrect index to be written into the manifold and sometimes it corrupts the array.

jrouwe, 2022-01-02:
Note for a repro, take the code from here: 

https://github.com/jrouwe/BulletPerformanceTest

And change 'for (int mq = 0; mq < 2; ++mq)' into 'int mq = 1;' and change 'for (int num_threads = 1; num_threads <= (int)thread::hardware_concurrency(); ++num_threads)' into 'int num_threads=8'.

erwincoumans, 2022-01-03:
Thanks for the report. Can you check if this PR helps?
https://github.com/bulletphysics/bullet3/pull/4119

jrouwe, 2022-01-03:
That fixes the issue, thanks!

closed 2022-01-03 by jrouwe