bulletphysics/bullet3 #4133: bug in changeDynamics function for joint manipulation

lewisboyd, 2022-01-11:
Hi, I think I have found a minor bug in the changeDynamics function when used to change joint position and velocity limit in the same call.

I've included some code below which I've tested on version 3.2.1

In the first case, trying to change a joint's position and velocity limits in a single function call only causes the velocity limit to be changed. 

In the second case, having separate calls for changing the position and velocity limits results in them both being enforced correctly.

p.s. thanks for all the work you've put into PyBullet! 

```
import pybullet
import pybullet_data
import pybullet_utils.bullet_client as bc

p = bc.BulletClient(connection_mode=pybullet.GUI)

p.setAdditionalSearchPath(pybullet_data.getDataPath())

kukaId = p.loadURDF("kuka_iiwa/model_free_base.urdf",[0,0,1], useFixedBase=True)

newUpperLim = 1
newVelLim = 1.0

# Case 1 - Velocity limit enfored but position limits ignored
p.changeDynamics(
    kukaId, 1, jointLowerLimit=-1, jointUpperLimit=newUpperLim,
    maxJointVelocity=newVelLim
)

# Case 2 - Velocity and position limits enforced
# p.changeDynamics(kukaId, 1, jointLowerLimit=-1, jointUpperLimit=newUpperLim)
# p.changeDynamics(kukaId, 1, maxJointVelocity=newVelLim)

p.setJointMotorControl2(kukaId, 1, p.POSITION_CONTROL, 5)

for i in range(500):
    jointVel = p.getJointState(kukaId, 1)[1]
    if jointVel > newVelLim:
        print("velocity exceeded", jointVel)
    _ = p.stepSimulation()

jointPos = p.getJointState(kukaId, 1)[0]
if jointPos > newUpperLim:
    print("upper limit exceeded", jointPos)
```

lewisboyd, 2022-01-11:
For some background as to why I was trying to limit the joint velocity via the changeDynamics function, I was hoping to use the setJointMotorControlArray function but it doesn't take a max velocity parameter like setJointMotorControl2.

erwincoumans, 2022-01-21:
Thanks for the report, it is likely an issue indeed. For now, please use your suggested workaround (only set one parameter at a time). Will get back if/when I have time (too many other projects going on).