bulletphysics/bullet3 #4136: C Api documentation / features

RobertLemmens, 2022-01-13:
Is there any documentation other then scouring through the pybullet code to see whats possible? I feel like its missing many features or acts very different from the regular C++ api. I'd like to set upAxis for heightfields for example, cant seem to do that in the C api.

erwincoumans, 2022-01-21:
The C API only exposes what we needed for our robotics learning research, not every single C++ feature.
The [PyBullet Quickstart Guide documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3) and the pybullet.c is the reference for the features.

closed 2022-01-21 by erwincoumans

RobertLemmens, 2022-01-21:
Gotcha, thanks for responding!