bulletphysics/bullet3 #4148: Default collision group filter mode not AND as expected

cgokmen, 2022-01-28:
The documentation seems to suggest that the default collision filtering mode will be B3_FILTER_GROUPAMASKB_AND_GROUPBMASKA

[image]

However, unless the parameter is explicitly set to be 0, the collision mode appears to be the OR mode (as can be seen in the below example). getPhysicsEngineParameters also does not report the value of this parameter. 

For now we are able to work around this by explicitly setting this mode (`p.setPhysicsEngineParameter(collisionFilterMode=0)`), but after looking through the source code for bullet3 we could not figure out where and how this is defaulting to the OR mode so I'm filing this issue in case you're able to figure it out.

Below is a minimal repro example (thanks @ChengshuLi). Note that there currently appears to be no example anywhere (not even on a Google search) of the use of this setCollisionFilterGroupMask function with a non-zero group ID (which disables _all_ collisions), so perhaps the below can be used for this purpose. 

```
import pybullet as p

p.connect(p.DIRECT)

collision_filter_and_mode = 0  # B3_FILTER_GROUPAMASKB_AND_GROUPBMASKA
collision_filter_or_mode = 1  # B3_FILTER_GROUPAMASKB_OR_GROUPBMASKA

# !!!!!!!!!!!!!! Uncomment this line to make balls 2, 3 not collide as expected. !!!!!!!!!!!!!!!
# p.setPhysicsEngineParameter(collisionFilterMode=collision_filter_and_mode)

visual_id = p.createVisualShape(p.GEOM_SPHERE, radius=1)
collision_id = p.createCollisionShape(p.GEOM_SPHERE, radius=1)

all_group = -1
default_group = 1 << 0
isolated_group = 1 << 6

# Ball 1 is in the default group
ball1 = p.createMultiBody(baseMass=1, baseCollisionShapeIndex=collision_id, baseVisualShapeIndex=visual_id)
p.setCollisionFilterGroupMask(ball1, -1, default_group, all_group) 

# Ball 2 is in a separate group on its own
ball2 = p.createMultiBody(baseMass=1, baseCollisionShapeIndex=collision_id, baseVisualShapeIndex=visual_id)
p.setCollisionFilterGroupMask(ball2, -1, isolated_group, all_group) 

# Ball 3 is in the ALL group, collisions disabled w/ ball 2's group
ball3 = p.createMultiBody(baseMass=1, baseCollisionShapeIndex=collision_id, baseVisualShapeIndex=visual_id)
p.setCollisionFilterGroupMask(ball3, -1, default_group, all_group & ~isolated_group) 

p.stepSimulation()
print(len(p.getContactPoints(ball1, ball2)) > 0)  # Prints True as expected.
print(len(p.getContactPoints(ball1, ball3)) > 0)  # Prints True as expected.
print(len(p.getContactPoints(ball2, ball3)) > 0)  # We expect this to print False, but it prints True.
```

Akshay6077, 2024-06-14:
Hi I am also facing g some issue in  this like my object has some collision issues did you figure this out

cgokmen, 2024-06-14:
The particular fix for this problem was calling `p.setPhysicsEngineParameter(collisionFilterMode=0)`

Akshay6077, 2024-06-17:
So just to confirm this will turn of the collisions or what??