bulletphysics/bullet3 #4153: Can I get the position of multiple objects at once?

xyjbaal, 2022-02-07:
your work is really great.

I'm using pybullet to simulate some object stacking, there be about 1000 objects.
I need to constantly monitor the position of each object, 
but getBasePositionAndOrientation() can only get the position of one object, and if I use a for() loop, it will be very slow. 
It takes about 5 minutes each time to get the current position of all the objects.

I would like to ask if there is a good solution for this?

erwincoumans, 2022-03-30:
We should add some batch query for this indeed, it hasn't been added.

closed 2022-03-30 by erwincoumans

VallesMarinerisExplorer, 2023-02-21:
I am interested in this as well. Is there a script you would recommend looking at for someone who might write a plugin?