bulletphysics/bullet3 #4154: Speed up simulation for robot use

aled96, 2022-02-07:
I am trying to simulate, several times, a push with a manipulator but I do not understand how to speed up the simulation in order to run several tests in, for example, 1 second.

This is the idea of the code

```
pybullet.connect(pybullet.DIRECT)
            
pybullet.setGravity(0, 0, -9.81)
pybullet.setTimeStep(0.0001)
pybullet.setRealTimeSimulation(0)

#Load robot and cube models
#Change dynamics of the object
pybullet.changeDynamics(self.cube,
                                -1,
                                mass = self.objectMass,
                                lateralFriction = 1,
                                spinningFriction = -1,
                                rollingFriction = -1,
                                linearDamping=0,
                                angularDamping=0.1,
                                restitution = 0.05)

# Inverse Kinematic
#Move arm
for i in range(self.n_joints):
            jointInfo = pybullet.getJointInfo(self.robot, i)
            qIndex = jointInfo[3]
            
            if qIndex > -1:
                pybullet.setJointMotorControl2(bodyIndex=self.robot,
                                    jointIndex=i,
                                    controlMode=pybullet.POSITION_CONTROL,
                                    targetPosition=joint_pos[qIndex-7],
                                    targetVelocity=0,
                                    force=500,
                                    positionGain=0.02,
                                    velocityGain=25)
            
for i in range(10000):
      pybullet.stepSimulation()

```

With this, I need 2-3 seconds to run a single simulation... I tried to change the simulation step but the arm starts to move randomly...
In addition, I don't know if I have also to change the dynamic parameters of the object and/or the params in the setJointMotorControl2, according to a change in the simulation step.

Thank you !
Can you help me?

erwincoumans, 2022-03-30:
Don't set friction to negative values. Also, the default timestep of 240Hz should be able to simulate an arm, and we often use 240Hz or 1000Hz. Using 10k Hz seems excessive, so the setup has some issue.

closed 2022-03-30 by erwincoumans

aled96, 2022-03-30:
In the last month I have changed the friction values to positive values. But I still have a problem with the understanding of the timestep. In fact, if I use 0.001 or below, the robot starts to move abruptly without reaching the point desired.