bulletphysics/bullet3 #4163: On the action offset/scale/min/max values assigned to humanoid3d joints in pybullet_deep_mimic_env.py

joonkyu4220, 2022-02-14:
Hello sir, I found a little peculiarity while exploring your PyBullet implementation of DeepMimic.

In pybullet_deep_mimic_env.py([Here](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/deep_mimic/env/pybullet_deep_mimic_env.py)), defining build_action_offset/ build_action_scale/build_action_bound_min/build_action_bound_max,
I noticed that the scale/min/max values assigned to the **right shoulder**(17\~20th elements in the lists) and **left shoulder**(31\~34th) are not symmetric, while, to my knowledge, there is no reason for them to be different.

_(Here I labeled the corresponding joint in the comment and changed the number of digits in the fractional parts by adding zeroes, for readability.)_
```
      out_scale = [
          0.20833333333333, 1.00000000000000, 1.00000000000000, 1.00000000000000, # chest
          0.25000000000000, 1.00000000000000, 1.00000000000000, 1.00000000000000, # neck

          0.12077294685990, 1.00000000000000, 1.00000000000000, 1.00000000000000, # rightHip
          0.15923566878900, # rightKnee
          0.15923566878900, 1.00000000000000, 1.00000000000000, 1.00000000000000, # rightAnkle
          0.07961783439400, 1.00000000000000, 1.00000000000000, 1.00000000000000, # rightShoulder!!!!!!!!!!!!
          0.15923566878900, # rightElbow

          0.12077294685900, 1.00000000000000, 1.00000000000000, 1.00000000000000, # leftHip
          0.15923566878900, # leftKnee
          0.15923566878900, 1.00000000000000, 1.00000000000000, 1.00000000000000, # leftAnkle
          0.10775862068900, 1.00000000000000, 1.00000000000000, 1.00000000000000, # leftShoulder!!!!!!!!!!!!
          0.15923566878900 # leftElbow
      ]
```

```
      out_min = [
          -4.79999999999, -1.00000000000, -1.00000000000, -1.00000000000, # chest
          -4.00000000000, -1.00000000000, -1.00000000000, -1.00000000000, # neck

          -7.77999999999, -1.00000000000, -1.00000000000, -1.00000000000, # rightHip
          -7.85000000000, # rightKnee
          -6.28000000000, -1.00000000000, -1.00000000000, -1.00000000000, # rightAnkle
          -12.5600000000, -1.00000000000, -1.00000000000, -1.00000000000, # rightShoulder!!!!!!!!!!!!
          -4.71000000000, # rightElbow

          -7.77999999900, -1.00000000000, -1.00000000000, -1.00000000000, # leftHip
          -7.85000000000, # leftKnee
          -6.28000000000, -1.00000000000, -1.00000000000, -1.00000000000, # leftAnkle
          -8.46000000000, -1.00000000000, -1.00000000000, -1.00000000000, # leftShoulder!!!!!!!!!!!!
          -4.71000000000 # leftElbow
      ]
```

```
      out_max = [
          4.799999999, 1.000000000, 1.000000000, 1.000000000, # chest
          4.000000000, 1.000000000, 1.000000000, 1.000000000, # neck

          8.779999999, 1.000000000, 1.000000000, 1.000000000, # rightHip
          4.710000000, # rightKnee
          6.280000000, 1.000000000, 1.000000000, 1.000000000, # rightAnkle
          12.56000000, 1.000000000, 1.000000000, 1.000000000, # rightShoulder!!!!!!!!!!!!
          7.850000000, # rightElbow

          8.779999999, 1.000000000, 1.000000000, 1.000000000, # leftHip
          4.710000000, # leftKnee
          6.280000000, 1.000000000, 1.000000000, 1.000000000, # leftAnkle
          10.10000000, 1.000000000, 1.000000000, 1.000000000, # leftShoulder!!!!!!!!!!!!
          7.850000000 # leftElbow
      ]
```

Is there a specific reasoning behind this?
Am I missing something here, or misunderstanding the purpose of these codes?

I sincerely appreciate your work and thank you in advance for your responses to my issue.

erwincoumans, 2022-03-16:
Thanks, likely a copy / paste bug in the data.

joonkyu4220, 2022-03-17:
Thank you for your reply.
I'll close this issue now.

closed 2022-03-17 by joonkyu4220