bulletphysics/bullet3 #4183: Bullet fails to build with new TBB (>= 2020.3)

thesamesam, 2022-03-01:
Originally reported downstream in [Gentoo](https://bugs.gentoo.org/820827):
```
/var/tmp/portage/sci-physics/bullet-3.17-r1/work/bullet3-3.17/src/LinearMath/btThreads.cpp:39:10: fatal error: tbb/task_scheduler_init.h: No such file or directory
   39 | #include <tbb/task_scheduler_init.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Newer TBB has dropped this header so Bullet can't be built against it.

erwincoumans, 2022-03-15:
Unless someone fixes it, we can just drop TBB support.

closed 2022-03-15 by erwincoumans

bogdanrusu, 2025-05-26:
OpenMP is not supported on Android by clang as far as I know so for Android we are left with 3 options: 

1. use an older TBB (not sure if it would compile to Android though)
2. implement some custom std::thread implementation (which I would avoid as takes quite some effort to do properly and Bullet integrates so nicely with TBB like parallelism libs) 
3. adapt the btTaskSchedulerTBB to use the newer TBB API as @erwincoumans  suggests (see below an example)

As I understood, TBB has no official Android support but they are making efforts so that it runs on Android and while there are no guarantees my guess is that it works pretty well (I also did some tests and worked fine with Bullet - haven't spent much time on it though and I did had to comment some assertions (in BulletDynamics/ConstraintSolver/btSequentialImpulseConstraintSolver.cpp and BulletDynamics/ConstraintSolver/btBatchedConstraints.cpp) but that might be because I did something wrong when adding objects to the multi-threaded simulation)

I managed to make Bullet work with TBB v2022.1.0-rc1 with the below implementation (worked well on both MacOS and Android in my small project). 

In Bullet/src/LinearMath/btThreads.cpp:

```
#if BT_USE_TBB && BT_THREADSAFE
///
/// btTaskSchedulerTBB -- wrapper around Intel Threaded Building Blocks task scheduler
///
class btTaskSchedulerTBB : public btITaskScheduler
{
	int m_numThreads;

public:
	btTaskSchedulerTBB() : btITaskScheduler("IntelTBB")
	{
		m_numThreads = 0;
        oneapi::tbb::global_control global_limit(oneapi::tbb::global_control::max_allowed_parallelism, oneapi::tbb::this_task_arena::max_concurrency());

	}

	virtual int getMaxNumThreads() const BT_OVERRIDE
	{
		return oneapi::tbb::this_task_arena::max_concurrency();
	}
	virtual int getNumThreads() const BT_OVERRIDE
	{
		return m_numThreads;
	}
	virtual void setNumThreads(int numThreads) BT_OVERRIDE
	{
		m_numThreads = (std::max)(1, (std::min)(int(BT_MAX_THREAD_COUNT), numThreads));

		// Set the maximum number of threads globally
    	oneapi::tbb::global_control global_limit(oneapi::tbb::global_control::max_allowed_parallelism, m_numThreads);

		m_savedThreadCounter = 0;
		if (m_isActive)
		{
			btResetThreadIndexCounter();
		}
	}
	struct ForBodyAdapter
	{
		const btIParallelForBody* mBody;

		ForBodyAdapter(const btIParallelForBody* body) : mBody(body) {}
		void operator()(const tbb::blocked_range<int>& range) const
		{
			BT_PROFILE("TBB_forJob");
			mBody->forLoop(range.begin(), range.end());
		}
	};
	virtual void parallelFor(int iBegin, int iEnd, int grainSize, const btIParallelForBody& body) BT_OVERRIDE
	{
		BT_PROFILE("parallelFor_TBB");
		ForBodyAdapter tbbBody(&body);
		btPushThreadsAreRunning();
		tbb::parallel_for(tbb::blocked_range<int>(iBegin, iEnd, grainSize),
						  tbbBody,
						  tbb::simple_partitioner());
		btPopThreadsAreRunning();
	}
	struct SumBodyAdapter
	{
		const btIParallelSumBody* mBody;
		btScalar mSum;

		SumBodyAdapter(const btIParallelSumBody* body) : mBody(body), mSum(btScalar(0)) {}
		SumBodyAdapter(const SumBodyAdapter& src, tbb::split) : mBody(src.mBody), mSum(btScalar(0)) {}
		void join(const SumBodyAdapter& src) { mSum += src.mSum; }
		void operator()(const tbb::blocked_range<int>& range)
		{
			BT_PROFILE("TBB_sumJob");
			mSum += mBody->sumLoop(range.begin(), range.end());
		}
	};
	virtual btScalar parallelSum(int iBegin, int iEnd, int grainSize, const btIParallelSumBody& body) BT_OVERRIDE
	{
		BT_PROFILE("parallelSum_TBB");
		SumBodyAdapter tbbBody(&body);
		btPushThreadsAreRunning();
		tbb::parallel_deterministic_reduce(tbb::blocked_range<int>(iBegin, iEnd, grainSize), tbbBody);
		btPopThreadsAreRunning();
		return tbbBody.mSum;
	}
};
#endif  // #if BT_USE_TBB && BT_THREADSAFE
```

Hope this helps someone. 
Thank you.