bulletphysics/bullet3 #4200: Heightfield replacement does not update its relative body's AABB

JewelryForge, 2022-03-14:
```python
shape_id = pybullet.createCollisionShape(pybullet.GEOM_HEIGHTFIELD, heightfieldData=height_field_data, **other_kwargs)
height_field_id = pybullet.createMultiBody(0, shape_id)
result1 = pybullet.getAABB(height_field_id, -1)
pybullet.createCollisionShape(pybullet.GEOM_HEIGHTFIELD, heightfieldData=another_height_field_data, replaceHeightfieldIndex=shape_id, **other_kwargs)
result2 = pybullet.getAABB(height_field_id, -1)  # result2 is the same as result1
```

erwincoumans, 2022-03-15:
You didn't call createMultiBody in your code.

closed 2022-03-15 by erwincoumans

JewelryForge, 2022-03-17:
> You didn't call createMultiBody a second time in your code.

Why should I call createMultiBody again? I don't want another height field.