bulletphysics/bullet3 #4202: Boxes placed on a kinematic/dynamic object move without external force application

asreejith-vcrs, 2022-03-14:
Hi,

(Bullet C++) When a bunch of stable arrangement of boxes are placed on a kinematic/dynamic body, without application of any external force except gravity, we would expect the boxes to stay as is, without any motion between any bodies. However, this is not the case in my application.

To reproduce this issue, I modified the `KinematicMultiBodyExample.cpp` such that the boxes rest on the rotating body. I then disabled rotation velocities to 0, so that the arrangement of boxes remain stable. However, on running the simulation, these boxes fall off over time.
Here is the diff for the changes I made to `KinematicMultiBodyExample.cpp` so that the boxes rest on the stationary body:

```
diff --git a/examples/MultiBody/KinematicMultiBodyExample.cpp b/examples/MultiBody/KinematicMultiBodyExample.cpp
index 5a81d40e4..b7ea6ac3c 100644
--- a/examples/MultiBody/KinematicMultiBodyExample.cpp
+++ b/examples/MultiBody/KinematicMultiBodyExample.cpp
@@ -17,9 +17,9 @@ subject to the following restrictions:
 //#define USE_MOTIONSTATE 1
 #include "btBulletDynamicsCommon.h"
 #include "LinearMath/btTransformUtil.h"
-#define ARRAY_SIZE_Y 5
-#define ARRAY_SIZE_X 5
-#define ARRAY_SIZE_Z 5
+#define ARRAY_SIZE_Y 4
+#define ARRAY_SIZE_X 4
+#define ARRAY_SIZE_Z 4
 
 #include "LinearMath/btVector3.h"
 #include "LinearMath/btAlignedObjectArray.h"
@@ -34,7 +34,7 @@ void kinematicPreTickCallback(btDynamicsWorld* world, btScalar deltaTime)
 	btMultiBody* groundBody = (btMultiBody*)world->getWorldUserInfo();
 	btTransform predictedTrans;
 	btVector3 linearVelocity(0, 0, 0);
-	btVector3 angularVelocity(0, 0.1, 0);
+	btVector3 angularVelocity(0, 0.0, 0);
 	btTransformUtil::integrateTransform(groundBody->getBaseWorldTransform(), linearVelocity, angularVelocity, deltaTime, predictedTrans);
 	groundBody->setBaseWorldTransform(predictedTrans);
 	groundBody->setBaseVel(linearVelocity);
@@ -44,7 +44,9 @@ void kinematicPreTickCallback(btDynamicsWorld* world, btScalar deltaTime)
 	time += deltaTime;
 	double old_joint_pos = groundBody->getJointPos(0);
 	double joint_pos = 0.5 * sin(time * 3.0 - 0.3);
+	joint_pos = 0.0;
 	double joint_vel = (joint_pos - old_joint_pos) / deltaTime;
+	joint_vel = 0.0;
 	groundBody->setJointPosMultiDof(0, &joint_pos);
 	groundBody->setJointVelMultiDof(0, &joint_vel);
 }
@@ -178,9 +180,9 @@ void KinematicMultiBodyExample::initPhysics()
 				for (int j = 0; j < ARRAY_SIZE_Z; j++)
 				{
 					startTransform.setOrigin(btVector3(
-						btScalar(0.2 * i),
-						btScalar(2 + .2 * k),
-						btScalar(0.2 * j)));
+						btScalar(-0.3 + 0.2 * i),
+						btScalar(0.7 + .2 * k),
+						btScalar(-0.3 + 0.2 * j)));
 
 					createRigidBody(mass, startTransform, colShape);
 				}

```

asreejith-vcrs, 2022-03-14:
This is how the example looks like:

https://user-images.githubusercontent.com/89802083/158256920-a3b5c6e9-90b7-46e8-82dd-9065d432964f.mp4

erwincoumans, 2022-03-15:
The example hasn't been tuned well, using a smaller time step (1./240.) and more solver iterations (50 for example) will reduce (but not eliminate) the drift.

closed 2022-03-15 by erwincoumans