bulletphysics/bullet3 #4208: CMake: modernize config file with imported targets & components

SpaceIm, 2022-03-16:
Current CMake config file installed by `bullet3` is a quite old and fragile way to design a config file.

It would be nice to properly export targets and create config file with built-in CMake functions (`configure_package_config_file`, `write_basic_package_version_file`, `install(EXPORT ...)`)

Imported targets could be:
- `Bullet::Bullet3Common`
- `Bullet::LinearMath`
- `Bullet::BulletInverseDynamics`
- `Bullet::BulletCollision`
- `Bullet::BulletDynamics`
- `Bullet::BulletSoftBody`
- etc
- and eventually an interface `Bullet::Bullet` target depending all components.

And usage would be something like this:

```cmake
find_package(Bullet REQUIRED BulletSoftBody CONFIG)
target_link_libraries(myApp PRIVATE Bullet::Bullet3Common Bullet::Bullet3Geometry Bullet::BulletSoftBody)
```

SpaceIm, 2022-03-16:
Actually, I see that it was added few years ago then reverted in https://github.com/bulletphysics/bullet3/pull/1303 :(

erwincoumans, 2022-04-14:
It was reverted since the new cmake broke various things (don't remember what exactly). I agree the build system and the code base should be improved, but we lack resources to work on this and rather keep it as-is.

closed 2022-04-14 by erwincoumans