bulletphysics/bullet3 #4221: Will getContactPoints return infinite points?

jingxixu, 2022-03-28:
Hi everyone, I am trying to understand the function [getContactPoints](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.cb0co8y2vuvc). If two solid bodies (meshes) penetrate, there should be an infinite number of contact points. **Imagine two triangular faces intersect at a line and the line will contain infinite number of points.** How does this function handle it so that it does not return infinite number of contact points? Thanks a lot!

xissburg, 2022-03-28:
There's a limit of four contact points, which define a contact polygon. As part of collision detection, there is an algorithm which chooses which points to keep as to maximize the area of the contact polygon.

jingxixu, 2022-03-28:
Just to confirm, is the contact polygon the intersection surface between the two rigid bodies, as shown in the figure below? Do you mind sharing a reference to the code/algorithm that you talked about? Many thanks!

[image]

xissburg, 2022-03-28:
Yes, pretty much. In your example, there should be 4 contact points, one at each vertex of the dashed rectangle. 
The [btPersistentManifold::addManifoldPoint](https://github.com/bulletphysics/bullet3/blob/22d9eec148d3a7dec9d7b10a4344808bf766f73a/src/BulletCollision/NarrowPhaseCollision/btPersistentManifold.cpp#L212) function will select which point to replace if the maximum count has been reached. `btPersistentManifold::sortCachedPoints` implements that logic using the `calcArea4Points` function to make the best choice..