bulletphysics/bullet3 #4240: Axis changed after saving urdf via urdfEditor if inertial frame has rotation.

TexnoMann, 2022-04-18:
I found a broken axis issue after using urdfEditor. The axis for wrist_3_joint broke after that I opened urdf with urdfEditor and saved it unchanged. I've noticed that this only happens for links that have a non-zero rotation for the child link's inertial frame. I have attached the file.

I also have a problem if the geometry for the collision is described as box, sphere or cylinder. The saved result will contain an unknown reference and give me an error:
```xml
<mesh filename="unknown_file" scale="1.00000 1.00000 1.00000"/>
```

TexnoMann, 2022-04-18:
#4231

erwincoumans, 2022-04-19:
Indeed, the urdfEditor is very limited and doesn't support all features. There isn't going to be development in this area, so closing for now. If someone has a contribution, please file a pull request.

closed 2022-04-19 by erwincoumans