bulletphysics/bullet3 #4245: _WIN32 in btVector3::maxDot()

1vanK, 2022-04-21:
https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btVector3.h#L1001-L1007
```
#if defined _WIN32 || defined(BT_USE_SSE)
	const long scalar_cutoff = 10;
	long _maxdot_large(const float* array, const float* vec, unsigned long array_count, float* dotOut);
#elif defined BT_USE_NEON
	const long scalar_cutoff = 4;
	extern long (*_maxdot_large)(const float* array, const float* vec, unsigned long array_count, float* dotOut);
#endif
```

Should it be without _WIN32 like

https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btVector3.h#L1033-L1044

```
SIMD_FORCE_INLINE long btVector3::minDot(const btVector3* array, long array_count, btScalar& dotOut) const
{
#if (defined BT_USE_SSE && defined BT_USE_SIMD_VECTOR3 && defined BT_USE_SSE_IN_API) || defined(BT_USE_NEON)
#if defined BT_USE_SSE
	const long scalar_cutoff = 10;
	long _mindot_large(const float* array, const float* vec, unsigned long array_count, float* dotOut);
#elif defined BT_USE_NEON
	const long scalar_cutoff = 4;
	extern long (*_mindot_large)(const float* array, const float* vec, unsigned long array_count, float* dotOut);
#else
#error unhandled arch!
#endif
```

erwincoumans, 2022-04-24:
We only tested it for Visual Studio/_WIN32, it may work with other compilers. Please try it.

closed 2022-04-24 by erwincoumans

1vanK, 2022-04-25:
We test this with many compilers https://github.com/urho3d/Urho3D/blob/master/Source/ThirdParty/Bullet/src/LinearMath/btVector3.h#L1004-L1008

1vanK, 2022-04-25:
Also SSE is enabled for MinGW https://github.com/urho3d/Urho3D/blob/master/Source/ThirdParty/Bullet/src/LinearMath/btScalar.h#L77-L81