bulletphysics/bullet3 #4260: pybullet can not show all objects.

JiayunjieJYJ, 2022-05-14:
Either DebugVisualizer or getCameraImage, some objects don't show when I import multiple URDFs.

JiayunjieJYJ, 2022-05-14:
[image]
For example, each robot is located under each green line, but only a few of them can be shown, anyone know the reason?

JiayunjieJYJ, 2022-05-14:
I have tried using different computers, including laptop with CPU or a PC with GPU, but when I import multiple URDFs, only top 8 or 8  of them can be shown, the left is just ignored.

erwincoumans, 2022-05-20:
The number of triangles is likely too high, please simplify the visual (and possibly collision) meshes.

closed 2022-05-20 by erwincoumans