bulletphysics/bullet3 #4265: Change the type of inertia to btmatrix3x3 to suit the general situation

microaaron, 2022-05-16:
[btVector3 m_invInertiaLocal;](https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Dynamics/btRigidBody.h#L69)
The inertia tensor should be represented by a 3x3 matrix.

erwincoumans, 2022-05-17:
It is not, since the rigid body center of mass frame is aligned with the principle axis of inertia and the center of mass. You can shift the collision and visual geometry the opposite way. This is done in PyBullet and the URDF importer.

closed 2022-05-17 by erwincoumans

microaaron, 2022-05-17:
Although we can shift the collision and visual geometry to make the rigid body center of mass frame aligned with the principle axis of inertia and the center of mass. i still hope to provide more options.