bulletphysics/bullet3 #4287: Using pybullet inside casadi for dynamics computation

prakrutk, 2022-06-17:
I am trying to use pybullet inside casadi for computing the next states but it gives a segmentation fault as I try to pass optimization variable (force value and the coordinates where force is to be applied) inside pybullet.applyExternalForce. 

I am not sure but I guess it is because casadi uses a symbolic framework and I cannot pass symbols in a pybullet function. If any one has a clearer idea on why this is happening and is there any way to get this working please do let me know.

Thanks.

YechenZhou, 2023-03-08:
Have you solved this problem yet? I have the same problem as you. @prakrutk

prakrutk, 2023-03-13:
I guess it is mostly due to the symbolic framework while pybullet does not take in symbolic variables in the function. I did not find any solution I used another physics engine in place of this to make it work.

closed 2023-03-13 by prakrutk

jcarpent, 2023-03-13:
Depending on your applications, you may use Pinocchio for that (https://github.com/stack-of-tasks/pinocchio/).

YechenZhou, 2023-03-13:
> I guess it is mostly due to the symbolic framework while pybullet does not take in symbolic variables in the function. I did not find any solution I used another physics engine in place of this to make it work.
Can you tell me what physics engine you used afterwards?

prakrutk, 2023-03-13:
> > I guess it is mostly due to the symbolic framework while pybullet does not take in symbolic variables in the function. I did not find any solution I used another physics engine in place of this to make it work.
> > Can you tell me what physics engine you used afterwards? @prakrutk

I used Mujoco for my application. But it depends on what you are using it for.

YechenZhou, 2023-03-13:
> I used Mujoco for my application. But it depends on what you are using it for.

 Thanks. Can Mujoco work inside casadi? @prakrutk