bulletphysics/bullet3 #4290: Visualizing multiple pyBullet server instances

manas96, 2022-06-21:
Hi, I'd like to spawn multiple instances of the physics server, send them some geometry, start them and then combine their output in a single GUI window. I'm trying to repurpose the existing 3D rendering infrastructure that pyBullet provides. Here's what I'm trying to do so far:
```
p0 = bc.BulletClient(connection_mode=pybullet.DIRECT)
p1 = bc.BulletClient(connection_mode=pybullet.DIRECT)
p0.loadURDF("r2d2.urdf")
p1.loadSDF("stadium.sdf")

# This will be used to render the output from p0 and p1 both
pgui = bc.BulletClient(connection_mode=pybullet.GUI)

while 1:
    p0.stepSimulation()
    p1.stepSimulation()
    
    # somehow get all geometry information

 pgui.stepSimulation()

```

manas96, 2022-07-06:
I was able to hack something together by maintaining two copies of geometry objects. That is, if we have geometries {X,Y} on p1, {A,B} on p2, I maintain geometries {X,Y,A,B} on the GUI server and update their positions every frame. 
But happy to hear if there is a cleaner solution.

yogha001, 2023-01-27:
Hi @manas96, I am currently working on the same issue and I was wondering if you have found a solution for it since then?
Or @erwincoumans  could you help us ?

manas96, 2023-01-30:
Hi @yogha001, I ended up building on top of the workaround I posted above.

yogha001, 2023-01-30:
Hi @manas96, so happy you replied. Well i'm not really working on excatly the same issue. 
Here is the issue I'm working on, just made a post: https://github.com/bulletphysics/bullet3/issues/4402