bulletphysics/bullet3 #4302: Visualization: training process is suspended!

newera-001, 2022-07-12:
Env:
  Ubuntu 20.04.3 LTS (GNU/Linux 5.13.0-48-generic x86_64)
  Pybullet : 3.1.8
  python : 3.7.0
  gym : 0.20.0

Description：
My RL Env is laikago env which copy from https://github.com/erwincoumans/motion_imitation. So my probelm is that when I visualize my trainning process and timesteps reach 6m~7m, my visualization would be suspended: the robot seems frozen and windows does not have any changes, but meanwhile process still  is alive.

Situation as below:
[image]

erwincoumans, 2022-07-19:
Have you check resources, in particular memory usage?

newera-001, 2022-08-01:
yes. the graphical memory, ram and cpu were used.
[image]
[image]