bulletphysics/bullet3 #4307: Observation space of InvertedPendulumBulletEnv-v0?

mhr, 2022-07-20:
I noticed that the inverted pendulum Gym environment (https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/gym_pendulum_envs.py) is 5-dimensional, which is different from the usual 4-dimensional observation vector: [cart position, cart velocity, pendulum angle, pendulum angular velocity]. What do the observation dimensions correspond to?

Thanks!

GHUserNameLL, 2022-07-22:
Hi, Same issue here. But BTW, do you know the meaning of action space in invertedPendulumBulletEnv-v0? I see the action space is 1-dimension, but I don't know if it means the pole's moving position or velocity?

mhr, 2022-07-22:
It's not the pole. It's the cart. The action space is torque (force) applied to the cart in either the right or left direction, one of which has a positive sign and the other negative. How much is the cart being pushed in one direction or the other? That's the scalar action.

GHUserNameLL, 2022-07-22:
> It's not the pole. It's the cart. The action space is torque (force) applied to the cart in either the right or left direction, one of which has a positive sign and the other negative. How much is the cart being pushed in one direction or the other? That's the scalar action.

Thanks!!! Your answer really helps me a lot!