bulletphysics/bullet3 #4322: Pybullet : Is vhacd the best for detail collision shape?

YuhyeonHwang, 2022-08-12:
Hi,

I have been setting up a contact-rich environment for reinforcement learning
For collision shape, obj file was loaded with a convex hull mesh using v-hacd (provided by pybullet)

But collision shape is different from original obj file, so there is a problem in performing contact-rich task.

Is there any other way for the best collision mesh load?

( Is it an inherent problem with the physical engine? I'm a beginner... )

thank you

hyuh0810, 2022-10-13:
This paper is good for me understand the above problem :
https://sites.google.com/nvidia.com/factory/

jadon995, 2022-10-24:
> 

Hi. I am having the same problem in simulating a contact-rich task. Do you have any solutions to this problem?

hyuh0810, 2023-01-18:
@jadon995 I've looked it up for a while, and here's my opinion.
If you want to load static parts, you can use triangle mesh
see [#2532](https://github.com/bulletphysics/bullet3/issues/2531)
also find this information in the pybullet documentation.

But if you want to load moving parts(like robot ee), you cannot use triangle meshes, but rather convex hulls. therefore I recommend using VHACD

baldr08, 2023-12-12:
hello,I face the problem that my robots endeffector cant touch completly with a curved face,as shown in the figrue
[image]
,I tried to add concave="yes" to link label of the urdf,but they still cant touch compeletly.Do you have any solutions to this problem?
thank you.