bulletphysics/bullet3 #4337: Segmentation fault when loading a UNITREE Go1 URDF

stephane-caron, 2022-09-08:
* PyBullet version: 3.2.5

PyBullet hits a segmentation fault when loading the UNITREE Go1 description from [``go1.urdf``](https://github.com/unitreerobotics/unitree_mujoco/blob/main/data/go1/urdf/go1.urdf):

```
pybullet build time: May 20 2022 19:44:17
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Intel
GL_RENDERER=Mesa Intel(R) HD Graphics 520 (SKL GT2)
GL_VERSION=4.6 (Core Profile) Mesa 21.2.6
GL_SHADING_LANGUAGE_VERSION=4.60
pthread_getconcurrency()=0
Version = 4.6 (Core Profile) Mesa 21.2.6
Vendor = Intel
Renderer = Mesa Intel(R) HD Graphics 520 (SKL GT2)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: FR_thigh_shoulder
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: FL_thigh_shoulder
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: RR_thigh_shoulder
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: RL_thigh_shoulder
[1]    8354 segmentation fault (core dumped)  python examples/show_in_pybullet.py go1
```

### How to reproduce

Run the PyBullet script from [`robot_descriptions.py`](https://github.com/stephane-caron/robot_descriptions.py) with the UNITREE Go1 description:

```console
python examples/show_in_pybullet.py go1
```

Alternatively, download the description to some local directory and run:

```python
    GO1_URDF_PATH = "/some/local/path/go1.urdf"
    client = pybullet.connect(pybullet.GUI)
    pybullet.configureDebugVisualizer(pybullet.COV_ENABLE_SHADOWS, 0)
    pybullet.configureDebugVisualizer(pybullet.COV_ENABLE_GUI, 0)
    pybullet.setAdditionalSearchPath(os.path.dirname(GO1_URDF_PATH))
    robot = pybullet.loadURDF(os.path.basename(GO1_URDF_PATH))
    input("Press Enter to close terminate... ")
    pybullet.disconnect()
```

yxyang, 2022-10-28:
Just encountered the same issue. Looks like removing the following line at the beginning of the URDF file would work:
```xml
<link name="world"/>
  <joint name="" type="floating">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="world"/>
    <child link="trunk"/>
  </joint>
```

learning-github-zgy, 2024-05-14:
> Just encountered the same issue. Looks like removing the following line at the beginning of the URDF file would work:刚刚遇到同样的问题。看起来删除 URDF 文件开头的以下行会起作用：
> 
> ```
> <link name="world"/>
>   <joint name="" type="floating">
>     <origin rpy="0 0 0" xyz="0 0 0"/>
>     <parent link="world"/>
>     <child link="trunk"/>
>   </joint>
> ```

Thank you, you have solved my problem, I had been stuck with this problem for days, thank you very much!
Do you know why you deleted this paragraph?