bulletphysics/bullet3 #4348: pybullet GUI

mysweetdcm, 2022-09-25:
I have a puzzle. I built a gym environment using Pybullet and turned on the Direct mode during training. Everything works fine. When I turn on p.Setrealtimesimulation (1) during testing, the performance of the agent is much worse than the training result. If p.timesimulation is not enabled, the agent behaves normally, but the time step is faster than the RealTime mode. I am very confused about this. Thank you very much for your reply

jjshoots, 2022-12-06:
If no one answered you yet, the reason for this is that `setRealTimeSimulation(1)` steps the environment as fast as possible to maintain real time visuals, this is sometimes a lot less than with `setRealTimeSimulation(0)`.

Due to this, the environment during evaluation and training ends up being very different due to different physics loop rates.

jjshoots, 2022-12-06:
See also [this](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.mq73m9o2gcpy)

erwincoumans, 2023-02-24:
SetRealTimeSimulation is meant only for GUI mode, so step the simulation based on the real-time clock. In Direct mode you are supposed to call the stepSimulation manually.

closed 2023-02-24 by erwincoumans