bulletphysics/bullet3 #4368: Cannot import pybullet_envs

ManosL, 2022-11-04:
Before starting I want to mention that I have install the OpenAI gym version `0.26.2` and pybullet version `3.2.5`. When I try to import `pybullet_envs` package, I get the following error:

```
Traceback (most recent call last):
  File "/home/manosl/Desktop/MSc_Courses/MSc_Thesis/Code/soft_actor_critic_implementation/main.py", line 1, in <module>
    import pybullet_envs
  File "/home/manosl/Desktop/MSc_Courses/MSc_Thesis/Code/thesis_env/lib/python3.10/site-packages/pybullet_envs/__init__.py", line 15, in <module>
    register(
  File "/home/manosl/Desktop/MSc_Courses/MSc_Thesis/Code/thesis_env/lib/python3.10/site-packages/pybullet_envs/__init__.py", line 7, in register
    if id in registry.env_specs:
AttributeError: 'dict' object has no attribute 'env_specs'

```

Does anyone knows how to fix this issue? I suspect the reason for this is the following:

The error is raised at pybullet_envs/__init__.py which does the following:

```
def register(id, *args, **kvargs):
  if id in registry.env_specs:
    return
  else:
    return gym.envs.registration.register(id, *args, **kvargs)
```

I changed it to this and the error was removed

```
def register(id, *args, **kvargs):
  if id in registry:
    return
  else:
    return gym.envs.registration.register(id, *args, **kvargs)
```

If you want you can have a look 
Thanks

zzf-zzf, 2022-12-05:
This resolution is not work for me, do you have any other method?

araffin, 2022-12-22:
You can patch gym as done in the RL Zoo: https://github.com/DLR-RM/rl-baselines3-zoo/pull/256

The file: https://github.com/DLR-RM/rl-baselines3-zoo/blob/feat/gym-0.24/rl_zoo3/gym_patches.py
and then use `apply_api_compatibility=True` when creating the env.

(soon you will be able to apply the patches by simply doing `import rl_zoo3.gym_patches`)

LJoson, 2022-12-27:
> This resolution is not work for me, do you have any other method?

when i change gym0.26.0 -> 0.18.0,it works

SiggyF, 2022-12-30:
This error is caused by [this](https://github.com/openai/gym/pull/3041) update to the gym environment (since 0.26), the `env_spec` was removed in favor of a simple dictionary. The suggested change looks good to me.

erwincoumans, 2023-02-24:
This requires a fix in Bullet? Is there a run-time version check of gym, so we can implement both old and new method?

araffin, 2023-02-27:
> This requires a fix in Bullet? Is there a run-time version check of gym, so we can implement both old and new method?

This PR should already fix it: https://github.com/bulletphysics/bullet3/pull/4332 (it should work with both gym 0.21 and 0.26)

The check in gym is here: https://github.com/openai/gym/blob/master/gym/envs/registration.py#L497-L498

araffin, 2023-11-08:
Fyi, I pushed and released on pypi a subset of pybullet envs compatible with gymnasium: https://github.com/araffin/pybullet_envs_gymnasium
`pip install pybullet_envs_gymnasium` (I will crosspost this message as it may interest several people)