bulletphysics/bullet3 #4369: Convert code to support Gymnasium instead of Gym

ManosL, 2022-11-05:
Considering that Gym is not maintained by OpenAI and its maintained now from a different repo [Gymnasium](https://github.com/Farama-Foundation/Gymnasium), I suggest to change the code in order to import `gymnasium` instead of `gym`, in order for the codebase to be up-to-date with the recent developments of Gym

Sitin, 2023-06-28:
Are there any plans to include this into the roadmap? I don't think that this will be too hard and I would love to help with any existing effort.

rickstaa, 2023-07-14:
> Are there any plans to include this in the roadmap? I don't think this will be too hard, and I would love to help with any existing effort.

I quickly checked if I could quickly migrate the examples to Gymnasium, although it is not hard I think the job takes quite an effort since not only the import has to be changed but also the:

- Step method.
- Reset method.
- Seeding behaviour.
- Rendering behaviour.

For more information, see https://gymnasium.farama.org/content/migration-guide. In the end, decide to use the compatibility wrappers provided by Farama for my research (see https://gymnasium.farama.org/content/gym_compatibility/).

```python
register(
    id="WrappedMinitaur",
    entry_point="pybullet_envs.bullet.minitaur_gym_env:MinitaurBulletEnv",
    disable_env_checker=True,
    apply_api_compatibility=True,
)
```

araffin, 2023-11-08:
Fyi, I pushed and released on pypi a subset of pybullet envs compatible with gymnasium: https://github.com/araffin/pybullet_envs_gymnasium
`pip install pybullet_envs_gymnasium` (I will crosspost this message as it may interest several people)