bulletphysics/bullet3 #4370: colors.png not found

mbatuhancelik, 2022-11-06:
I installed the pybullet using pip and when I try to run something I get the following error:
I believe the mentioned file should be under data folder of the repo but I could not find it.

Any ideas on the solution?

```
EGL device choice: -1 of 4.
Loaded EGL 1.5 after reload.
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=NVIDIA GeForce RTX 2060/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 470.141.03
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA
Version = 4.6.0 NVIDIA 470.141.03
Vendor = NVIDIA Corporation
Renderer = NVIDIA GeForce RTX 2060/PCIe/SSE2
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [ur10e/visual/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [ur10e/visual/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [ur10e/visual/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [ur10e/visual/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
[... 146 lines cut ...]
not found [./data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [ur10e/visual/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [./data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../data/colors.png]
b3Warning[examples/Importers/ImportMeshUtility/b3ImportMeshUtility.cpp,182]:
not found [../../../../data/colors.png]
Destroy EGL OpenGL window.
```

erwincoumans, 2022-11-08:
>> when I try to run something 

What did you run exactly?

closed 2022-11-08 by erwincoumans

mbatuhancelik, 2022-11-09:
> > > when I try to run something
> 
> What did you run exactly?
I tried to import ur10 from urdf file, but I guess the file is lacking in the robot simulator files.
Sorry for the inconvenience.