bulletphysics/bullet3 #4380: CMake prints wrong order of linked libraries by target_link_libraries

LilyWangLL, 2022-12-07:
There are some issues about the link libraries order. Use incorrect order in target_link_libraries, it will cause linker errors like the description in https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=9554. And the similar issue: https://pybullet.org/Bullet/phpBB3/viewtopic.php?p=18132&hilit=undefined+reference+to+btCollisionObject#p18132, https://pybullet.org/Bullet/phpBB3/viewtopic.php?p=42962&hilit=undefined+reference+to+btCollisionObject#p42962, https://github.com/bulletphysics/bullet3/issues/3167. 
The correct order should be `BulletDynamics BulletCollision BulletSoftBody LinearMath`.
 Is this a bug of bullet3? If it is, could someone can help solve it?

dg0yt, 2022-12-07:
> CMake prints wrong order of linked libraries by target_link_libraries

It is not Cmake which prints libraries, but vcpkg, based on a text file.
And the "libraries" actually are targets which are exported from cmake, based on patching in vcpkg.

However, it would be usefull if there was official *exported* cmake config, instead of the simple variable-based file. In particular, vcpkg wants to support multi-config cmake generators.

erwincoumans, 2023-02-24:
This seems a vcpkg issue, not Bullet/cmake?

closed 2023-02-24 by erwincoumans