bulletphysics/bullet3 #4385: VSCode Terminal Shutdowns Example Browser

Razgaleh, 2022-12-23:
I tried to run this simple code in vscode terminal and the pop up example browser shutdowns immediately. I don't know what is causing this to happen. I looked around and didn't see similar problem I was wondering if anyone can help. The code runs perfectly in anaconda environment.  But I cannot run in the terminal - here's the proof that it shutdowns immediately: 

`
import pybullet as p

p.connect(p.GUI)
p.setPhysicsEngineParameter(enableFileCaching=0)
plane_shape = p.createCollisionShape(p.GEOM_PLANE)
floor = p.createMultiBody(plane_shape,plane_shape)
box_shape = p.createCollisionShape(p.GEOM_BOX, halfExtents = [1, 1, 1])
box_object1 = p.createMultiBody(box_shape,box_shape)
box_object2 = p.createMultiBody(box_shape,box_shape)
p.setGravity(0,0,-10)

`
Here's what the console shows: 

pybullet build time: Dec 22 2022 20:13:02
starting thread 0
started testThreads thread 0 with threadHandle 0000000000000484
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.0 NVIDIA 527.56
Vendor = NVIDIA Corporation
Renderer = NVIDIA GeForce GTX 1080/PCIe/SSE2
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 00000000000006F8
MotionThreadFunc thread started
numActiveThreads = 0
**stopping threads
Thread with taskId 0 with handle 00000000000006F8 exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 0000000000000484 exiting
Thread TERMINATED**

avikde, 2023-12-12:
Are you using a Mac by any chance?

I don't have a solution but I started running into this issue after switching to a Mac (M2). I am using it in a C++ program linking to bullet version 3.24. It does not exhibit this issue on Windows.

ramtinfarajollahi, 2023-12-14:
I had the same issue
Try running your script using
python -i 'youFIlePathName'

aleantli, 2024-06-07:
@ramtinfarajollahi : Ran into same issue and your solution worked. Thank you so much!