bulletphysics/bullet3 #4386: URDF visual is not shown after removing and re-creating the object several times

alper111, 2022-12-29:
**What I am doing:**
I am repeating the following cycle: (1) load an object from a URDF file, (2) interact with it with a robot, (3) then delete it.
**Problem:**
After iteration number 125 (after creating and removing the object 125 times), the object visual is not shown. However, I can see the collision shape by pressing `w`, and I can interact with the object with the cursor.
**No problem if:**
There is no problem if I load the URDF once and reset its position.
**Expectation:**
The visual of the object would not be affected by the number of load/removal times.
**Recording of the mentioned effect:** https://youtu.be/4bPfYZ6zjK8 
**Sample code:**
```python
import os
import time

import pybullet as p
import pybullet_data

p.connect(p.GUI)
p.configureDebugVisualizer(p.COV_ENABLE_GUI, 0)
p.resetDebugVisualizerCamera(1.5, 0, -40, [0.55, -0.35, 0.2])

robot = p.loadURDF(os.path.join(pybullet_data.getDataPath(), "franka_panda/panda.urdf"), useFixedBase=True)
table = p.loadURDF(os.path.join(pybullet_data.getDataPath(), "plane_transparent.urdf"))

it = 0
while True:
    it += 1
    p.resetJointState(robot, 0, 0)
    p.resetJointState(robot, 1, 0.307)
    p.resetJointState(robot, 2, 0)
    p.resetJointState(robot, 3, -2.7)
    p.resetJointState(robot, 4, 0)
    p.resetJointState(robot, 5, 3)
    p.resetJointState(robot, 6, -2.356)
    p.resetJointState(robot, 9, 0.01)
    p.resetJointState(robot, 10, 0.01)

    obj = p.loadURDF(os.path.join(pybullet_data.getDataPath(), "soccerball.urdf"), [0.6, 0., 0.05], globalScaling=0.1)
    # p.stepSimulation()

    joints = p.calculateInverseKinematics(robot, 11, [0.4, 0., 0.05])
    for _ in range(50):
        p.setJointMotorControlArray(robot, [0, 1, 2, 3, 4, 5, 6], p.POSITION_CONTROL, targetPositions=joints[:-2])
        p.stepSimulation()
        if it > 120:
            time.sleep(1/240)
    joints = p.calculateInverseKinematics(robot, 11, [0.6, 0., 0.05])
    for _ in range(50):
        p.setJointMotorControlArray(robot, [0, 1, 2, 3, 4, 5, 6], p.POSITION_CONTROL, targetPositions=joints[:-2])
        p.stepSimulation()
        if it > 120:
            time.sleep(1/240)
    p.removeBody(obj)
    print(it)
```

ManuCorrea, 2022-12-29:
I can confirm I also get this error running the script, still see the collision shape pressing w.

erwincoumans, 2023-02-24:
Eventually the GPU/OpenGL will run out of memory, we don't handle this case. Try running separate instances of the simulation in different threads/processes instead.

closed 2023-02-24 by erwincoumans

ManuCorrea, 2023-02-25:
I dived deeper into the problem and to my findings instead of using the information already existing about the object in the buffer with every iteration the buffer kept growing. I couldn't really clear myself if it was growing out of adding vertex information, some indexing error or some metadata of each object. It was some time ago and let it on the side but I could come back to it and provide more detailed information.

But for this case the warning Buffer Limit would help the user to know.